CeuAzul/ADR #27: Static margin is fucking crazy as fuck

rafaellehmkuhl, 2019-02-11:
@davidssand please post the static margin crazy plots here

davidssand, 2019-02-11:
[image]
Static margin still not plotting well. Should be constant, around 0,14.

rafaellehmkuhl, 2019-02-14:
[image]
 UUUUUUUUUUUUUUUUUUUUUUUUUUUU NIIIIIIIIIIIIIIIIIIIIIIIIIIIICEEEE

Problem whas using CM instead of dCM/dAlpha on the static margin calculation.

rafaellehmkuhl, 2019-02-14:
Biplane is also beautiful and matches EESC values (from presentation: 0-50%)

[image]

rafaellehmkuhl, 2019-02-14:
For reference, this is the CM_alpha for the biplane (Micro2018 dimensions and wings separated by 60cm):

[image]

PR #34 (2019-02-14): Big change, fixing Flight_stability
- StaticMargin fixed #27 
- Flight_stability name is now a class (FlightStability) #33 
- test_FlightStability created #33 
- Plane class can support 2 wings and has method for printing plane, CG and CA 
- Class arguments are now a dictionary.

[image]

[image]
closed 2019-02-14 by rafaellehmkuhl